frtz13/UPSPlus_mqtt #2: TimeoutError on i2c bus crashed fanShutDownups ...

ArjenR49, 2021-05-20:
From the cronlog on my UPS Plus running your scripts I see the same error that I experienced with the GeeekPi scripts (see below).
I combatted them with 'except TimeoutError:' in the scripts I wrote based on the the GeeekPi stuff and having very little experience with python spent a lot of time finding out about exception handling only to find failing writes which weren't exceptions .... I did get over those, too. Rereading and rewriting until OK ... Something perhaps with memory management of the MCU, as Ed Watson suggested. The write problems concerned the 'countdown' registers...

The TimoutError in the log below seems to have crashed fanShutDownups.py .... I can't find it it HTOP any longer ...

UPS Plus to MQTT  Copyright (C) 2021  https://github.com/frtz13
This program comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under conditions of the GPL (see http://www.gnu.org/licenses for details).

Type ctrl-C to exit
Traceback (most recent call last):
  File "/home/pi/scripts/fanShutDownUps.py", line 530, in <module>
    handleUPS(mqttclient)
  File "/home/pi/scripts/fanShutDownUps.py", line 348, in handleUPS
    upsPlus = C_UPSPlus(upsCurrent)
  File "/home/pi/scripts/fanShutDownUps.py", line 226, in __init__
    self.aReceiveBuf.append(i2c_bus.read_byte_data(DEVICE_ADDR, i))
  File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 433, in read_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
TimeoutError: [Errno 110] Connection timed out

frtz13, 2021-05-20:
added exception handling to the reading of the reading of the UPS registers.
that said, I have been running this script for a couple of weeks now, and it never crashed for this reason. did you give your UPS a reset by taking out the batteries ?

ArjenR49, 2021-05-20:
Not today, but when I updated the f/w, of course. That's a week or so ago.
I'll take the batteries out today, but I think I already had these TimeoutError exceptions early on, on version 5 and before. I even wrote test scripts to get an idea of how often they occur. There are so many variables and not having made extensive written notes, I can't be 100% sure.
A day or two ago I found that writing to countdown registers often failed when the f/w was  actually counting down.
GeekPi's idea was to set the countdown at 0x1A to 120 and every minute upsPlus.py would set it back to 120. I used 180, but evry now and then the write to 180 would fail. In the end I won by reading back what was written and if not correct, write again ... and again until OK. Writing a 0 seemed to be no problem, perhaps because it stops the counting down. So then I started to set 0x1A to 0 before setting it back to 180.
This was to implement a watchdog function, btw. The idea seems fine, and it worked when it worked. However, when writes were missed, the Pi would be stopped and never restart. Stuff like that.

Perhaps my UPS board is faulty to a degree. Something marginal in the design? Or the MCU's memory handling is 'lazy', a lazy software lock as Ed Watson suggested. I can only imagine, what it could be ...

I am now running your latest version ...

ArjenR49, 2021-05-20:
I get the error message:
mei 21 01:02:04 RPI-HUB /fanShutDownUps.py[419]: [handleUPS] Error getting data from UPS: [C_UPSPLus.init] Error reading UPS registers: [Errno 110] Connection timed out
mei 21 01:02:04 RPI-HUB kernel: i2c-bcm2835 fe804000.i2c: i2c transfer timed out

about once every few hours, but they're handled nicely and the control script goes on running and doing its business.

closed 2021-05-20 by ArjenR49