frtz13/UPSPlus_mqtt #3: Spelling error corrections?

ArjenR49, 2021-05-20:
Can I just somehow upload the script to this site with 2 spelling error corrections I made? I have never done a thing like that, but I slowly realize, how dd-wrt development tracking is done. I've followed that for years ... Apparently the old and the new files are compared and the changes marked automatically. I always somehow thought that the developer marks the changes by hand ... simple minded old fashioned programmer, started with Algol60 as a course task and wrote mainly 'plain and simple' SAS code for years.

ArjenR49, 2021-05-20:
Thank you and let's see ... I did try to read through the instructions about Pull Requests. Very long and detailed indeed. No short cuts for beginners.
It'll have to wait. I rather much doubt I can contribute to the code.
I could rewrite all the register addresses in hexadecimal (as in EP-136 document) and bit-wise shifts into octal ... but that's maybe not what you'd like ;-)

frtz13, 2021-05-20:
how about just telling me about the spelling errors so I can fix these?

ArjenR49, 2021-05-20:
Sure, maybe leave studying the Pull Request stuff for more serious purposes ;-)

So, please look in fanShutDownups.py for volage and intial.

Should be voltage and initialization, of course.

frtz13, 2021-05-20:
ok, thanks.

closed 2021-05-21 by ArjenR49