frtz13/UPSPlus_mqtt #7: Maybe wrong way on crontab ?!

hellresistor, 2021-05-24:
Hello friend.

This weekend i have tested your repository.
Are work great! :) except the crontab reboot command :P Will reboot the rpi after some time ;)

frtz13, 2021-05-24:
the crontab line by itself won't reboot the RPi. it starts a task at boot time.
the fanShutDownUps.py task never reboots your RPi, it can only shut it down when battery is low (that's its main purpose). if this task shuts down your RPi , it will tell you about this fact in syslog. if this happened unexpectedly, have a look at your syslog.

hellresistor, 2021-05-24:
Sorry. Just Now i have noticed. It is the script itself. 

The syslog

```
~# tail -f /var/log/syslog

May 24 19:59:49 raspberrypi CRON[5242]: (CRON) info (No MTA installed, discarding output)
May 24 20:00:01 raspberrypi CRON[5251]: (root) CMD (/usr/bin/python3 /root/bin/upsPlus_iot.py)
May 24 20:00:17 raspberrypi CRON[5250]: (CRON) info (No MTA installed, discarding output)
May 24 20:00:26 raspberrypi systemd[1]: Started Session 11 of user pi.
May 24 20:00:36 raspberrypi crontab[5287]: (root) BEGIN EDIT (root)
May 24 20:00:44 raspberrypi 344f662187e3[818]:
May 24 20:00:47 raspberrypi crontab[5287]: (root) REPLACE (root)
May 24 20:00:47 raspberrypi crontab[5287]: (root) END EDIT (root)
May 24 20:00:52 raspberrypi /fanShutDownUps.py: UPS on battery. Battery voltage: 3.340 V. Shutdown at 3.900 V
May 24 20:00:52 raspberrypi /fanShutDownUps.py: UPS battery voltage below threshold of 3.900 V. Shutting down.
May 24 20:00:54 raspberrypi systemd[1]: Unmounting RPC Pipe File System...
May 24 20:00:54 raspberrypi systemd[1]: Stopping Session 11 of user pi.
May 24 20:00:54 raspberrypi systemd[1]: Stopping Virtual Machine and Container Registration Service...
```

frtz13, 2021-05-24:
this looks as it should. UPS on battery, battery voltage under threshold = battery_protection_voltage + margin = 3700 + 200 mV.
hence shutdown.
shutdown does not happen at low battery condition when the battery is charging.

hellresistor, 2021-05-24:
I think see this issue on Upsplus repository

https://github.com/geeekpi/upsplus/issues/36

closed 2021-05-24 by hellresistor

frtz13, 2021-05-24:
are you saying that the RPi shut down, while the charger was connected and charging the batteries ?

hellresistor, 2021-05-24:
yes. And seems the UPS  stops charging batteries (randomly). I am waiting to get some info, if is a hardware problem or a firmware issue.

deHarro, 2021-05-25:
@frtz13 
I did some tests on that issue, so I can step in and answer your question.

> that the RPi shut down, while the charger was connected and charging the batteries

Not completely true.
The charger is attached and powers the UPS.
But there is no current flow from the USB port, where the charger is attached, to neither the batteries nor the PRi.
The RPi in this situation is powered only by the batteries and the batteries are not charged nor backed by the charger.

Additionally we have a further problem in this situation: 
The four blue LEDs signal FULL state (all four statically ON), despite the fact, that the batteries are depleted to a much lower than sane end of discharge voltage. My batteries show around 2,5 V in this situation.

> shutdown does not happen at low battery condition when the battery is charging.

This is another misinterpretation.
As far as the program flow of the original GeeekPi script "upsPlus.py" is concerned, you are right.
But this script misses the situation in discussion.
It assumes "charging the batteries" as long as there is any voltage over 4 V at one of the USB input ports.
If so, it doesn't check for depleted batteries.
In our error case, there is voltage at the charger port, but the batteries are nevertheless depleting.

Harry

frtz13, 2021-05-25:
> It assumes "charging the batteries" as long as there is any voltage over 4 V at one of the USB input ports.
> If so, it doesn't check for depleted batteries.
regarding the script in this repository, charging/discharging state is derived from average battery current. the script will shut down the RPi when battery voltage is under threshold and avg. battery discharge current is > 500mA.

deHarro, 2021-05-25:
Upps! Sorry!
I missed the topic of this issue.
My elaboration on the additional problem with the LEDs signaling wrong status is targeted on the original script code provided by GeeekPi (where hellresistor posted, too). Same for the "misinterpretation" part above. I should have emphasized more precisely, that "this script" means the original.

Besides of that...
If I had realized earlier, that your repo contains code for the UPS, fitting for non MQTT, too, I would have tested it.
But now it's too late, I just sent back my two UPS to GeeekPi to give them the chance to track down the error, hellresistor and I encounter from time to time with these devices.

Harry

hellresistor, 2021-05-25:
@frtz13 
add this to the repository. for the lazy ^^

```
binary_sensor:
  - platform: mqtt
    name: "UPS on Battery"
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OnBattery"] }}'
    payload_on: "True"
    payload_off: "False"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Battery Charging"
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryCharging"] }}'
    payload_on: "True"
    payload_off: "False"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
        
sensor:
  - platform: mqtt
    name: "UPS average battery current"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryCurrent_avg_mA"] }}'
    unit_of_measurement: "mA"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "UPS Battery Voltage"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryVoltage_V"] }}'
    unit_of_measurement: "V"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Current Battery"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryCurrent_mA"] }}'
    unit_of_measurement: "mA"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Battery Power"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryPower_avg_mW"] }}'
    unit_of_measurement: "mW"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Battery Capacity"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryRemainingCapacity_percent"] }}'
    unit_of_measurement: "%"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Battery Temperature"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["BatteryTemperature_degC"] }}'
    unit_of_measurement: "Cº"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Output Voltage"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputVoltage_V"] }}'
    unit_of_measurement: "V"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "OutPut Voltage Min"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputVoltage_mini_V"] }}'
    unit_of_measurement: "V"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Output Current"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputCurrent_mA"] }}'
    unit_of_measurement: "mA"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Output Current avg"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputCurrent_avg_mA"] }}'
    unit_of_measurement: "mA"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Output Power avg"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputPower_avg_mW"] }}'
    unit_of_measurement: "mW"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"
  - platform: mqtt
    name: "Output Current Peak"
    device_class: current
    state_topic: "home/rpi/ups"
    value_template: '{{ value_json["OutputCurrent_peak_mA"] }}'
    unit_of_measurement: "mA"
    availability:
      - topic:  "home/rpi/LWT"
        payload_available: "online"
        payload_not_available: "offline"

```

deHarro, 2021-05-25:
> @deharo , Try this, Works! ;)

Hi @hellresistor ,

perhaps you better use @deHarro to address me? ;-)

But at the moment I am without UPS hardware. I sent them back to GeeekPi, they want to check those two UPS.

Harry

frtz13, 2021-05-26:
@hellresistor : uploaded sensor definitions