frtz13/UPSPlus_mqtt #8: HA_*.yaml file is not in Unicode UTF-8

ArjenR49, 2021-05-27:
When I open the downloaded zip file, the *.yaml file is not UTF-8.
I checked with file -i after a text editor complained. I think it was Geany that allowed me to convert the file to UTF-8.
That doesn't mean I got it to work, because I don't properly know what to do with it. Where to put it etc. Instructions abound on the internet, but most are concerned with stand-alone installations (I mean without docker/IOTstack).
The fact, in my case at least, that the container data files in IOTstack/volumes are owned by root, is a surprise. I'm not sure it is as it is supposed to be since I have seen no mention of it and it complicates things as the OS won't allow the file manager to be root. You have to use a terminal, which is cumbersome.

frtz13, 2021-05-27:
the HA_config_ups_fan.yaml file is not meant to be used directly. it contains sensor definitions which you can copy into the Home Assistant config.yaml file.
something else: I have the impression you installed the Home Assistant container in IOT stack. please note that this does not yield in a "Home Assistant supervised" installation. see https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/
a HA supervised installation lets you add the Mosquitto MQTT broker add-on (and a file editor add-on to edit config.yaml etc.). after that you can add some of the sensor definitions to your config.yaml file and finally add the new sensors to your Lovelace GUI.

ArjenR49, 2021-05-28:
I demolished docker (https://lokarithm.com/2020/05/31/how-to-completely-remove-docker-from-debian-ubuntu-or-your-raspberry-pi/) and deleted all IOTstack files to possibly start anew. This is a test installation on a different SSD than I use for this machine otherwise (which has MotionEye on Raspberry OS). I could also wipe the whole SSD if that would help. Installing your scripts is straightforward, easy to do. They are great.

"I have the impression you installed the Home Assistant container in IOT stack. please note that this does not yield in a "Home Assistant supervised" installation. see https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/"
The words 'yield a ...' throw me off.

I had HA in an IOTstack until this morning. I had done what Andreas Spiess recommended. I have now removed it all, but when I then read the instructions in the link above, there's Docker and IOTstack again ... and about Home Assistant Supervised  I read so many warnings about it being very complicated to maintain Debian and HA, that I better stay far from it ...

Comes to mind what I decided back in the 60's ... I am not going to use another shitty Belling & Lee coax connector for my ham radio stuff, but standardise on BNC, even though I was just a sec. school student. Same here, I decided to standardise on Raspberry Pi and its own OS and not look at Banana Pi, Orange Pi or Odroid. I'd just loose my way (and money) in that scenery.

The only way to avoid Docker, seems to be installing HA as an operating system, but that would surely mean I cannot run MotionEye on the same machine (MEYE would also prefer to be its own OS ...)

My UPS's task is to power a Pi with both VNC server and client(viewer) and MotionEye and what have you, MuttMail etc. so I need Raspberry OS.

I could install HA and MQTT on my DNS server Pi which has PiHole, Unbound and Wireguard on Raspberry Pi OS, but only if it doesn't interfere with what that server is doing now, providing DNS for my LAN in any way. Probably not too busy doing that.

This HA stuff started looking way to complicated for me to solve in a decent amount of time.

frtz13, 2021-05-28:
I have been running this "HA supervised" installation on Raspberry Pi OS (Buster) for a couple of months now. sure, this is an "unsupported" installation.

closed 2021-05-29 by ArjenR49