frtz13/UPSPlus_mqtt #9: How to change entity name?

ArjenR49, 2021-05-31:
I noticed you have changed at least a lot of names in the yaml file. I have now managed to install HA and Mosquitto on my Pi with the help of hellresistor, which was great, although a lot of problems along the way.

However, he used an old version of the yaml file (HA configuration).
In my experience, if one changes the name of an entity in that file only, it will become unavailable in the HA dashboard.
Where else would one have to change the name to make the entity work again?

If I replace the old yaml file with your latest one, in which many names have changed for the better, will they show up after restarting everything? With just one name change a reboot didn't help at all. I waited overnight ... still unavailable. 

I also changed the units of voltage, current and power to V, A and W, and that was very succesfull thanks to a good example of formatting in the files (concurrent changes to the py script and the yaml file)!

frtz13, 2021-05-31:
in each sensor definition, the name: parameter defines the name of the sensor in Home Assistant. btw I add UPS... to the sensor names so it is easy to select the in lists. just begin typing UPS and you'll have all the UPS sensors at hand.
I suggest the following debugging procedure:
have a look at your MQTT traffic. Home Assistant / configuration / integrations / Mosquitto broker / configure / listen to a topic. Enter home/rpi/# / start listening (use whatever topic you defined in your fanShutDownUps.ini file).
you should immediately get a retained home/rpi/LWT message that the UPS is online. you should then get an MQTT message form home/rpi/ups containing a JSON payload every minute. initially, Check the fields in the payload against the fields defined in your sensor definitions in configuration.yaml, specifically the value_template definitions.
Further, the states of your sensors should be visible in Home Asssistant / Developer Tools / States (scroll down to sensor.ups.....)
depending on the sensor names you'll find here you may need to rework the sensor definitions for the HA dashboard.

ArjenR49, 2021-06-02:
Thanks! I hadn't guessed that the fancy name is not just a fancy name, but sets the sensor's name to some extent. Adding a question mark to the fancy name doesn't change the sensor's name, nor does capitalizing, but changing words around certainly changes the sensors name (entity). Once I had noticed that, things got better. I deleted the widget with all the old entities  and built a new widget.
By the time I had set it all up to my satisfaction and was ready to take a rest from HA ;-) GeeekPi decided to put a new f/w (8) up for us ...
The rest is history ...
I had to drop the HA project/SSD and went to just running your script and MotionEye on the UPS/Pi (different SSD) and take care not to cause any writing to i2c because that seems to set off a flurry of i2c timeout errors.
The problems have been reported and I'll just wait until the next f/w version for now before going back to the HA installation. I was already planning of moving HA/MQTT from the UPS/Pi to a different Pi, but it'll have to wait.