frtz13/UPSPlus_mqtt #13: Fan control experience

ArjenR49, 2021-07-22:
I finally found myself in a position to add the (very simple) hardware required for PWM control of the ICE Tower fan that I use with the Pi4 on the UPS+ ('with the fancy lights' by Geeekpi; 2-wire, no PWM), so I could enable PWM control in frtz13's script. I chose GPIO18, hardware PWM, for the control (to base of transistor).

The default settings of the PI control algorithm seem to work well with the ICE Tower. It has a very large mass compared to the CPU and therefore a large delay, but the PI control looks like it's doing fine.
The control hardware is very simple: a BJT (2N2222A) with grounded emitter, a series base resistor of 1 kOhm, and a Schottky flywheel diode, which is rumoured to come already included in most brushless fans, so perhaps superfluous (not including it will save a wire connection ...).

Even with a stress test running all 4 CPU's at over 90% the ICE Tower keeps the CPU at ca. 55 degrees Celsius (as set in the INI file) on a duty cycle of only about 50% (35% ... 70%).

The same PI control loop didn't do too well on a different PI4 with only heatsinks (and a PoE). In this case the fan was not provided by GeeekPi, but by Noctua (not a PWM model, either).

frtz13, 2021-07-30:
thanks for the feedback.

> The same PI control loop didn't do too well on a different PI4 (...)

what happened?

ArjenR49, 2021-08-01:
Sorry, I am working on three Pi4 servers and being slightly chaotic (by my own admission) I have no recollection. I have no sure fire way of setting parameters Kp and Ki. (I have renamed PI control parameters to be able to easier compare script code to sources describing PID controllers)

All in all, I have come to the conclusion that the best behaving small fans are Noctua 40x40x20 mm PWM fans. They can be controlled without problems, without level converter (e.g. directly from GPIO 18) and without producing extra noise because of the PWM frequency. I will not be able to repair those three (eventually five) servers, if needed, for months in a row, which is the reason I go for Noctua fans. Two (eventually 4) servers run 24/7 at 100% CPU load.

I got some cheap thin 40x40 fans which came with an IcyBox 4 level 'tower' case and while they surprisingly don't make noise to speak of in the new state, the cooling properties seem lacking in my setup with a PoE hat between fan and Pi, and they won't run at below 50% duty cycle. I have yet to see how long they last. The Noctua fans seem to deplace the air more effectively, and maybe they can run faster, I don't know.

Reliability starts to weigh more than fan speed control in my view. PWM control starts looking more like a PI(D) learning exercise than something totally essential.
RPM measuring would be worthwhile adding, though, allowing to remotely check if the fans are actually running, whether PWM controlled or not.

closed 2021-08-14 by ArjenR49