frtz13/UPSPlus_mqtt #14: "Cannot initialize communication with INA219 (output)"

hellresistor, 2021-07-28:
heya!
Some ideas for this error?

```
root@rpi4-20210718:~# python3 UPSPlus_mqtt/fanShutDownUps.py
UPS-Plus to MQTT version 20210624 Copyright (C) 2021  https://github.com/frtz13
This program comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under conditions of the GPL (see http://www.gnu.org/licenses for details).

Type ctrl-C to exit
Cannot initialize communication with INA219 (output). Error message: Could not determine default I2C bus for platform.

```

frtz13, 2021-07-28:
just a guess: try to add the busnum parameter in line 626, to make it look like:
inaRPi = INA219(0.00725, busnum=DEVICE_BUS, address=0x40)
can't check it myself (currently away from home).
if ok, the same in line 635.

hellresistor, 2021-07-28:
I have tryed with on the last post situation.

https://github.com/geeekpi/upsplus/issues/39#issuecomment-886225573

and continue getting  issue. (error respectivly)

hellresistor, 2021-07-28:
after some minutes running. not sure but maybe (15m/30m)

```
Type ctrl-C to exit
[handleUPS] Error getting data from UPS: [C_UPSPLus.init] Error reading UPS registers: [Errno 121] Remote I/O error

```

hellresistor, 2021-07-30:
Maybe solved add it on other 3rd place.

```
try:
        void = i2c_bus.read_byte_data(DEVICE_ADD,  0x12)
```
Testing..

closed 2021-07-31 by hellresistor