frtz13/UPSPlus_mqtt #24: HA and MQTTbroker addon implement

hellresistor, 2021-12-12:
Heya friend.

Is some issue when I use HA MQTT Addon and sync with your script? :|

I am able to get info using a local mqtt server (has you show before).

any tip?

frtz13, 2021-12-12:
no issues as far as I know. the script has been running like this for months now. as far as I know, you have to use login credentials for the mosquitto broker; the current version does not support anonymous connections any more.

hellresistor, 2021-12-12:
yes i have put right password. but not Discovering :|
I am putting the Topic "home/rpi" on MQTT Addon to Listen. Nothing happens.. :\ what I missing?! :O 

[image]

mqtt config
[image]
[image]

frtz13, 2021-12-12:
listen to this topic: home/rpi/#
once you see the mqtt messages, define the Sensors in configuration.yaml as stated in readme.

hellresistor, 2021-12-12:
you refer on .ini file? or addon mqtt?

frtz13, 2021-12-12:
home assistant / configuration / integrations / mosquitto broker, configure / listen to topic, subscribe to: home/rpi/#

hellresistor, 2021-12-12:
nothing .... 

MQTT Addon LOG after start the python. this tell you something?

```
[17:20:39] INFO: Successfully send discovery information to Home Assistant.
[17:20:39] INFO: Successfully send service information to the Supervisor.
1639329638: Config loaded from /etc/mosquitto/mosquitto.conf.
1639329638: Loading plugin: /usr/share/mosquitto/auth-plug.so
1639329638:  ├── Username/password checking enabled.
1639329638:  ├── TLS-PSK checking enabled.
1639329638:  └── Extended authentication not enabled.
1639329638: Opening ipv4 listen socket on port 1883.
1639329638: Opening ipv6 listen socket on port 1883.
1639329638: Opening websockets listen socket on port 1884.
1639329638: Warning: Mosquitto should not be run as root/administrator.
1639329638: mosquitto version 1.6.12 running
1639329638: New connection from 127.0.0.1 on port 1883.
1639329638: Socket error on client <unknown>, disconnecting.
1639329664: New connection from 172.30.32.1 on port 1883.
1639329664: New client connected from 172.30.32.1 as 6P7Tur7dnyBdgmgRhUtiEj (p2, c1, k60, u'fifu').
1639331010: New connection from 192.168.1.191 on port 1883.
1639331010: Socket error on client <unknown>, disconnecting.
```

hellresistor, 2021-12-12:
DONE! DETECTING. have change the name to IP

frtz13, 2021-12-12:
in Home Asssistant / Supervisor / Mosquitto broker / Configuration: chapter Logins:
declared a username + password (see Mosquitto broker documentation).
configured this username/password in fanShutdownUps.ini ?
in INI file: configured: broker=localhost ?

hellresistor, 2021-12-13:
well since 10AM i start get this errors on HA Core LOGs  (keep eye on time)

```
File "/usr/src/homeassistant/homeassistant/core.py", line 912, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.ups_iot_platform_reply. State max length is 255 characters.

2021-12-13 10:09:19 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'home/rpi/ups': '{"UsbC_V": 5.02, "UsbMicro_V": 0.016, "OnBattery": false, "BatteryVoltage_V": 4.2, "BatteryCurrent_mA": -55, "BatteryCurrent_avg_mA": 32, "BatteryPower_avg_mW": 129, "BatteryCharging": false, "BatteryRemainingCapacity_percent": 98, "BatteryTemperature_degC": 40, "OutputVoltage_V": 4.896, "OutputVoltage_mini_V": 4.888, "OutputCurrent_mA": 1403, "OutputCurrent_avg_mA": 1340, "OutputPower_avg_mW": 6563, "OutputCurrent_peak_mA": 1488, "UPSPlus_IOT_Platform_Reply": "sending UPS status data failed: HTTPSConnectionPool(host='api.52pi.com', port=443): Max retries exceeded with url: /feed (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xffffa9a2c820>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 249, in message_received
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 629, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1222, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 912, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.ups_iot_platform_reply. State max length is 255 characters.

2021-12-13 10:10:30 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'home/rpi/ups': '{"UsbC_V": 5.036, "UsbMicro_V": 0.0, "OnBattery": false, "BatteryVoltage_V": 4.2, "BatteryCurrent_mA": -11, "BatteryCurrent_avg_mA": 23, "BatteryPower_avg_mW": 91, "BatteryCharging": false, "BatteryRemainingCapacity_percent": 98, "BatteryTemperature_degC": 38, "OutputVoltage_V": 4.896, "OutputVoltage_mini_V": 4.888, "OutputCurrent_mA": 1313, "OutputCurrent_avg_mA": 1343, "OutputPower_avg_mW": 6578, "OutputCurrent_peak_mA": 1425, "UPSPlus_IOT_Platform_Reply": "sending UPS status data failed: HTTPSConnectionPool(host='api.52pi.com', port=443): Max retries exceeded with url: /feed (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xffffa9a2c820>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 249, in message_received
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 629, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1222, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 912, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.ups_iot_platform_reply. State max length is 255 characters.

2021-12-13 10:12:42 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'home/rpi/ups': '{"UsbC_V": 4.997, "UsbMicro_V": 0.0, "OnBattery": false, "BatteryVoltage_V": 4.196, "BatteryCurrent_mA": -40, "BatteryCurrent_avg_mA": 28, "BatteryPower_avg_mW": 109, "BatteryCharging": false, "BatteryRemainingCapacity_percent": 98, "BatteryTemperature_degC": 39, "OutputVoltage_V": 4.896, "OutputVoltage_mini_V": 4.888, "OutputCurrent_mA": 1317, "OutputCurrent_avg_mA": 1345, "OutputPower_avg_mW": 6588, "OutputCurrent_peak_mA": 1475, "UPSPlus_IOT_Platform_Reply": "sending UPS status data failed: HTTPSConnectionPool(host='api.52pi.com', port=443): Max retries exceeded with url: /feed (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xffffa9a2c820>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 249, in message_received
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 629, in _async_write_ha_state
    self.hass.states.async_set(
[... cut ...]
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 629, in _async_write_ha_state
    self.hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1222, in async_set
    state = State(
  File "/usr/src/homeassistant/homeassistant/core.py", line 912, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.ups_iot_platform_reply. State max length is 255 characters.

```

frtz13, 2021-12-13:
comment out the Sensor definition for: UPS IOT Platform reply. you will still have this information in an attribute. or use a template to limit the length of this Sensor to 255 char.

closed 2021-12-14 by frtz13