frtz13/UPSPlus_mqtt #27: fanShutDownUPS.py forces a shautdown after 1hour Operation

barsailer, 2025-03-26:
Hi frtz13,
Your script is just great and allows to take advantage of the 52PI UPS Hardware.

I just wonder what is the aim of one_hour_delay variable onto line 681 (if upsplus.on_battery or (one_hour_delay == 0):) of fanShutDownUPS.py.

I do assume was inserted for test purposes, and I considered such "or" test condition shall be removed as it otherwises forces a shutdown after 3600s execution time.

Regards

frtz13, 2025-03-27:
Hi barsailer. the one_hour_delay allows to check for the following condition: battery voltage goes low, while having the charger connected. this happened once in my UPS due to maybe a hardware failure. I reflowed some components with hot air on the bord, it did not happen again so far. It is the SHUTDOWN_IMMEDIATELY_WHEN_ON_BATTERY parameter which I use to check the shutdown sequence and timing. otherwise you have to wait for the batterie voltage to go down. however, you are right, to be able to use this in normal operating conditions, you need to add the upsplus.on_battery condition in line 696. However, I'd rather recommend to use a relatively high protection voltage value, such as 4000 mV.

barsailer, 2025-03-27:
Dear, Frtz

 

Thanks so much for your quick response time and very valuable explanations.

 

In my operating mode, I already combined a High voltage protection as you also recommended.

 

Very strange that you’re getting « battery voltage goes low, while having the charger connected », this sounds to as irrationnal to me as the fact this UPS never stops Charging the Battery and 4th led keeps blinking with AC charger always ON.

 

I however I have to admit, I havent done objective Ic current measurement outside of INA219 measurements.

 

I would be curious getting this unit electrical diagrams, but could not find them on the net.

 

Outside of this never stop blinking concern, the 52PI HW together with your distribution, does the JOB perfectly, just crossing the fingers for long terms reliability 

Best Regards

barsailer, 2025-03-28:
Hi frtz13,

What a coincidence, after 30H test duration, I also experienced Battery going low while the charger is cnnected.

My UPS Plus is New Unit with firmware version 10, and I used your distribution with SHUTDOWN_IMMEDIATELY_WHEN_ON_BATTERY set to 1, Protection voltage at 4000mV  and the « or (one_hour_delay == 0) » test removed.

Analyzing syslogs, I got Error message « [UPS health] Charger connected, but battery is discharging » reported along last10 Hours before Raspberry died without any record in sysllog,  The battery sunk down under the protection voltage.

I’m actually recharging the batteries Off line and will proceed with further testing with your suggestion having = SHUTDOWN_IMMEDIATELY_WHEN_ON_BATTERY = 0 

Is it this ERROR message you caught and never happen again afetr your changes « [UPS health] Charger connected, but battery is discharging » ?

Regards

frtz13, 2025-03-28:
at that time I my script did not detect this error condition, so my RPi died when the UPS shut down due to low batteries, the RPi not being shut down properly. At that time I also observed that the blue LEDs did not always work as expected. For this reason I reflowed part of the board with my SMD rework hot air gun. however, this did not solve the blue LEDs problem completely.

frtz13, 2025-04-02:
I updated the script.

closed 2025-04-02 by frtz13

barsailer, 2025-04-03:
Thanks a lot frtz13,

 

I have home made version of your 20250402 distribution under test for some days already, indeed it fixes the unexpected Automatic shutdown issue, This is just Perfect.

 

Unfortunatly, The unit I have is having spurious inconsitent values returned by I2C measurements ( I assume) as follow.

*	[UPS health] Charger connected, but battery is discharging
*	UPS on battery. Battery voltage: 4.168 V. Shutdown at 4.000 V followed by UPS back on AC supply.
*	Protection voltage retrieved from UPS seems to have an incorrect value (4056 mV). Assumed to be 3500 mV

 

I’m currenly investigating possible root causes and conditions of occurrences suspecting an impact of the temperature conditions. I queried electrical schematic to 52PI the designer and I’m crossing the fingers 😉

 

In the mean time I decided to turn the IMMEDIATE_SHUTDOWN ON BATTERY option OFF, to not impact the RPI acitivities.

Thanks a lot for having updated your sources.